node-red/linux-installers #4: Node-Red not fully installed

agentflippy5, 2020-03-05:
I run the install script with this command:
`bash <(curl -sL https://raw.githubusercontent.com/node-red/linux-installers/master/deb/update-nodejs-and-nodered)`

It seemed to run successfully, but node-red will not start (see picture below)
[image]

I am running Ubuntu 18.04 on Windows Subsystem for Linux, and have installed build-essential

knolleary, 2020-03-05:
Have you looked in the error log it highlights?

In general, we prefer to handle support type questions on the forum - https://discourse.nodered.org - where the wider community are able to help.

dceejay, 2020-07-12:
should now be fixed with later patches

closed 2020-07-12 by dceejay