node-red/linux-installers #7: Install script fails on Raspbian buster (Failed to install Node.js )

Norde, 2020-07-16:
OS : openHABian 2.5.6 (Raspbian buster)
Package update : 16-07-2020

This occur at stage "Install Node.js" :

```
  Stop Node-RED                            ✔
  Remove old version of Node-RED ✔
  Remove old version of Node.js       -
  Install Node.js                               ✘   Failed to install Node.js - Exit
  Clean npm cache
  Install Node-RED core
  Move global nodes to local
  Install extra Pi nodes
  Npm rebuild existing nodes
  Add shortcut commands
  Update systemd script

```

/var/log/nodered-install.log give me no clue :
```
Started : Thu 16 Jul 2020 05:58:19 PM CEST
Running for user openhabian at /home/openhabian
Failed to stop nodered.service: Unit nodered.service not loaded.
Found global nodes:   :

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Reading package lists...
Building dependency tree...
Reading state information...
Package 'nodered' is not installed, so not removed
0 upgraded, 0 newly installed, 0 to remove and 5 not upgraded.
Already have nodejs
Now install nodejs
Versions: node: npm:
```

Tell me if you have any suggestion to improve my bug report

dceejay, 2020-07-16:
Do the OpenHAB guys support using the default script ? (I've never tried it)
What does `which node && node -v` report on command line  and `which npm && npm -v` also ?

Norde, 2020-07-17:
Yes, they support default script :

Official installation use GUI interface calling this "update-nodejs-and-nodered" script
https://github.com/openhab/openhabian/blob/master/functions/nodejs-apps.bash

Anyway same problem happen when directly using install script (bash <(curl -sL https://raw.githubusercontent.com/node-red/linux-installers/master/deb/update-nodejs-and-nodered))

node and npm command fail with a "Illegal instruction".
It seems that the installed versions are not compatible with arm6l.

I've checked latest node.js unoficial builds (https://unofficial-builds.nodejs.org/download/release/v12.18.2/) and it works well with a manual installation (both npm and node works).

dceejay, 2020-07-17:
did you move an sd card from an arm7 device down to an arm6 ?

(we point at 12.16.3 as that also includes x86 builds whereas the 12,18 doesn't yet.)

Norde, 2020-07-17:
Nope it's a fresh install.
I'll give a try with a clean raspbian to see if the problem is openhabian specific.

Norde, 2020-07-17:
It works on raspbian.
I'm sorry for disturbance this is a openhabian issue.

dceejay, 2020-07-18:
if their image has a pre-installed version of node.js then I would bet its the arm7 version .

Norde, 2020-07-18:
True, they recently switched to a APT based install.
However, raspbian repositories only have an armhf version.

But there is another problem because even uninstalling the preinstalled version of node-js, the script fails.

I opened a ticket at openhabian github, thanks for the follow-up;)

dceejay, 2020-07-18:
i'll also see if I can force arm6 version if needs be.
Try latest script version

Norde, 2020-07-18:
Latest script works fine !
Thanks a lot Dave

closed 2020-07-20 by dceejay