node-red/linux-installers #10: "No need for sudo" is misleading

njh, 2020-11-02:
Hi,

I have been trying to install node-red as a 'system' user - with uid `node-red`, rather than my personal user.

When running as root, you get presented a message saying:
```
Root user detected. Typically install as a normal user. No need for sudo.

Are you really sure you want to install as root ? (y/N) ? 
```

But then when running as the user `node-red`, I get the message:

```
We trust you have received the usual lecture from the local System
Administrator. It usually boils down to these three things:

    #1) Respect the privacy of others.
    #2) Think before you type.
    #3) With great power comes great responsibility.

[sudo] password for node-red: 
```

I think the message means you don't have to run `node-red` command with `sudo` but that isn't very clear.

dceejay, 2020-11-03:
The first message is ours - advising that generally it's not a good idea to install as root...
The second is from debian the first time you do something using sudo - so not something we can change.

Apologies I'm not quite clear on what your question/suggestion is ?

njh, 2020-11-06:
Hi Dave!

Sorry for not being clear.

When I first read the _"No need for sudo."_ message, I thought it meant that there was no need for `sudo` in order to perform the install. But I think what it actually means, is once it is installed the non-root user doesn't need sudo to run `node-red`.

My suggestion is to amend the message to something like:

```
Root user detected. Typically install as a normal user.
After installation there is no need for sudo to run node-red.

Are you really sure you want to install as root ? (y/N) ? 
```

nick.

dceejay, 2020-11-07:
well I would argue if you are installing as root you then are expecting to run as root... it's more to alert the user to the fact they don't need to install as root (using sudo) - as the script can't tell the difference I think between being run as the user root or a user with sudo... If we could we could just exit instead of letting them do it wrong.

knolleary, 2020-11-07:
Would it be clearer if the message said:

> You are running this script as the root user. Typical installs should be done as a regular user.
> If you are running this script using sudo, please cancel and rerun without sudo.
> If you know what you are doing as root, please continue.

njh, 2020-11-07:
@knolleary sounds great, very clear 👍

dceejay, 2020-11-07:
Have just found the $SUDO_USER variable... only set if running as sudo - so we can warn and fail instantly instead. (But if really are root and not using sudo it's not set so a real root user can install just fine...)

dceejay, 2021-04-15:
check added to script

closed 2021-04-15 by dceejay