node-red/linux-installers #11: NodeRed install script fails for ubuntu 20.10 desktop on raspberry pi

votecoffee, 2020-12-09:
### Which node are you reporting an issue on?
Node-Red install script: https://raw.githubusercontent.com/node-red/linux-installers/master/deb/update-nodejs-and-nodered

### What are the steps to reproduce?
Using image ubuntu-20.10-preinstalled-desktop-arm64+raspi.img.xz
sudo apt update
sudo apt upgrade
sudo apt full-upgrade

sudo rpi-eeprom-update
#CURRENT: Thu 03 Sep 2020 12:11:43 PM UTC (1599135103)

#misc
sudo adp intsall -y curl

#Python3
sudo apt install -y python3 python3-dev
sudo apt install -y python3-pip

#NodeRED
#https://nodered.org/docs/getting-started/raspberrypi
sudo apt install -y build-essential git
#bash <(curl -sL https://raw.githubusercontent.com/node-red/linux-installers/master/deb/update-nodejs-and-nodered)

### What happens?
Receive the error folder not found
Installation aborts/does not complete

### What do you expect to happen?
NodeRed installs

### Please tell us about your environment:

- [x] Node-RED version: 1.2.6
- [x] node.js version: 14.15.1
- [x] npm version: 6.4.8
- [x] Platform/OS: Raspberry Pi 4Model B  4GB firmware version 1599135103, ubuntu-20.10-preinstalled-desktop-arm64+raspi.img.xz
- [x] Browser: FireFox

knolleary, 2020-12-09:
Can you share the log with the error message to put it into context?

votecoffee, 2020-12-09:
I tried to recreate it in a fresh install as I did some things afterwards but the issue did not reproduce.  I also installed some additional libraries this time around, so not sure if those patched something in the background.  I'll pull the log from the previous failed attempt in case it's helpful.  It'll be about an hour as I'm in the middle of a couple of other things.

dceejay, 2020-12-10:
I have updated the script to attempt to create a symlink (will fail if already exists so no problem for existing systems).
Will leave open if you do manage to upload your logs as this may not be the actual issue you are reporting here.

votecoffee, 2020-12-10:
I figured out what I did.  I was in terminal in a build directory.  I had another terminal open where I deleted the working directory of the NodeRed install terminal.  So when I ran the script I got FileNotFoundError: [Errno 2] No such file or directory because the directory I was in became invalid.  It had nothing to do with the build scripts.

Sorry for the rookie mistake!

closed 2020-12-10 by votecoffee