node-red/linux-installers #12: Remove Python2 dependencies

Summer88, 2021-01-22:
I've running latest raspberry pi os, installed nodered by apt, then found that is needed Python2.
 
`Linux pi 5.4.83-v7+ node-red/node-red#1379 SMP Mon Dec 14 13:08:57 GMT 2020 armv7l GNU/Linux`

This are the packages that requires python2: 
```

  libbrotli1 libc-ares2 libjs-inherits libjs-is-typedarray libnode-dev libnode64 libpython-stdlib libpython2-stdlib libpython2.7-minimal libpython2.7-stdlib libssl-dev libuv1 libuv1-dev node-abbrev
  node-ajv node-ansi node-ansi-align node-ansi-regex node-ansi-styles node-ansistyles node-aproba node-archy node-are-we-there-yet node-asn1 node-assert-plus node-asynckit node-aws-sign2 node-aws4
  node-balanced-match node-bcrypt-pbkdf node-bluebird node-boxen node-brace-expansion node-builtin-modules node-builtins node-cacache node-call-limit node-camelcase node-caseless node-chalk
  node-chownr node-cli-boxes node-cliui node-clone node-co node-color-convert node-color-name node-combined-stream node-concat-map node-concat-stream node-config-chain node-console-control-strings
  node-copy-concurrently node-core-util-is node-cross-spawn node-cyclist node-dashdash node-decamelize node-decompress-response node-deep-extend node-defaults node-delayed-stream node-delegates
  node-detect-indent node-detect-newline node-duplexer3 node-duplexify node-ecc-jsbn node-editor node-encoding node-end-of-stream node-errno node-escape-string-regexp node-execa node-extend
  node-extsprintf node-find-up node-flush-write-stream node-forever-agent node-form-data node-from2 node-fs-vacuum node-fs-write-stream-atomic node-fs.realpath node-gauge node-get-caller-file
  node-get-stream node-getpass node-glob node-got node-graceful-fs node-har-schema node-har-validator node-has-flag node-has-symbol-support-x node-has-to-string-tag-x node-has-unicode
  node-hosted-git-info node-http-signature node-iconv-lite node-iferr node-import-lazy node-imurmurhash node-inflight node-inherits node-ini node-invert-kv node-is-builtin-module node-is-npm
  node-is-object node-is-plain-obj node-is-retry-allowed node-is-stream node-is-typedarray node-isarray node-isexe node-isstream node-isurl node-jsbn node-json-parse-better-errors node-json-schema
  node-json-stable-stringify node-json-stringify-safe node-jsonify node-jsonparse node-jsonstream node-jsprim node-latest-version node-lazy-property node-lcid node-libnpx node-locate-path
  node-lockfile node-lowercase-keys node-lru-cache node-mem node-mime-types node-mimic-fn node-mimic-response node-minimatch node-minimist node-mississippi node-mkdirp node-move-concurrently
  node-mute-stream node-node-uuid node-nopt node-normalize-package-data node-npm-package-arg node-npm-run-path node-npmlog node-oauth-sign node-object-assign node-once node-opener node-os-locale
  node-osenv node-p-cancelable node-p-finally node-p-limit node-p-locate node-p-timeout node-package-json node-parallel-transform node-path-exists node-path-is-absolute node-path-is-inside
  node-performance-now node-prepend-http node-process-nextick-args node-promise-inflight node-promzard node-proto-list node-prr node-pump node-pumpify node-punycode node-qs node-qw node-rc node-read
  node-read-package-json node-readable-stream node-registry-auth-token node-registry-url node-request node-require-directory node-require-main-filename node-resolve-from node-retry node-rimraf
  node-run-queue node-safe-buffer node-semver node-semver-diff node-set-blocking node-sha node-shebang-command node-shebang-regex node-signal-exit node-slash node-slide node-sorted-object
  node-spdx-correct node-spdx-expression-parse node-spdx-license-ids node-sshpk node-ssri node-stream-each node-stream-iterate node-stream-shift node-string-decoder node-string-width node-strip-ansi
  node-strip-eof node-strip-json-comments node-supports-color node-tar node-term-size node-text-table node-through node-through2 node-timed-out node-tough-cookie node-tunnel-agent node-tweetnacl
  node-typedarray node-uid-number node-unique-filename node-unpipe node-url-parse-lax node-url-to-options node-util-deprecate node-uuid node-validate-npm-package-license
  node-validate-npm-package-name node-verror node-wcwidth.js node-which node-which-module node-wide-align node-widest-line node-wrap-ansi node-wrappy node-write-file-atomic node-xdg-basedir
  node-xtend node-y18n node-yallist node-yargs node-yargs-parser nodejs python-minimal python2-minimal python2.7 python2.7-minimal

```

Python 2 should be or replaced by 3 https://pythonclock.org/

dceejay, 2021-01-22:
that is a Debian issue.

Use our command-line installer instead as per the docs and it should remove all the apt installed nodejs.
see
https://nodered.org/docs/getting-started/raspberrypi

Summer88, 2021-01-26:
Thank you, I've tried the script it seems it leave the system as it is.
I needed to remove the python2 then run it again.

> Running Node-RED install for user pi at /home/pi on raspbian
> 
> This can take 20-30 minutes on the slower Pi versions - please wait.
>   Stop Node-RED                       ✔
>   Remove old version of Node-RED      ✔
>   Remove old version of Node.js       -
>   Leave existing Node.js              -   Node v10.23.1   Npm 5.8.0
>   Clean npm cache                     -
> npm WARN npm npm does not support Node.js v10.23.1
> npm WARN npm You should probably upgrade to a newer version of node as we
> npm WARN npm can't make any promises that npm will work with this version.
> npm WARN npm Supported releases of Node.js are the latest release of 4, 6, 7, 8, 9.
> npm WARN npm You can find the latest version at https://nodejs.org/
>   Install Node-RED core               ✔   1.2.7 
> npm WARN npm npm does not support Node.js v10.23.1
> npm WARN npm You should probably upgrade to a newer version of node as we
> npm WARN npm can't make any promises that npm will work with this version.
> npm WARN npm Supported releases of Node.js are the latest release of 4, 6, 7, 8, 9.
> npm WARN npm You can find the latest version at https://nodejs.org/
>   Move global nodes to local          ✔
>   Install extra Pi nodes              ✔
>   Npm rebuild existing nodes          -
>   Add shortcut commands               ✔
>   Update systemd script               ✔
> 
> All done.
>   You can now start Node-RED with the command  node-red-start
>   or using the icon under   Menu / Programming / Node-RED
> 
> 

then:
` apt remove python2`

then:

> 
> This can take 20-30 minutes on the slower Pi versions - please wait.
> 
>   Stop Node-RED                       ✔
>   Remove old version of Node-RED      ✔
>   Remove old version of Node.js       ✔
>   Install Node.js LTS                 ✔   Node v12.20.1   Npm 6.14.10
>   Clean npm cache                     ✔
>   Install Node-RED core               ✔   1.2.7 
>   Move global nodes to local          -
>   Install extra Pi nodes              ✔
>   Npm rebuild existing nodes          ✔
>   Add shortcut commands               ✔
>   Update systemd script               ✔
> 
> 
> 
> All done.
>   You can now start Node-RED with the command  node-red-start
>   or using the icon under   Menu / Programming / Node-RED
>

dceejay, 2021-01-26:
we don't specifically remove python2 - we should remove all those node.js packages some of which pre-req python2 .

dceejay, 2021-11-17:
Time has moved on ... bullseye version of raspios now only has python3 etc

closed 2021-11-17 by dceejay