node-red/linux-installers #16: node-red-start leaks terminal window title

ygoe, 2021-07-10:
### What are the steps to reproduce?

* Install Node-RED as per [this guide](https://nodered.org/docs/getting-started/raspberrypi)
* Start it from the terminal by executing `node-red-start`
* Watch the log as it starts
* Quit the log view and get back to the shell prompt by pressing Ctrl+C

### What happens?

The terminal window title remains "Node-RED console" and doesn't change back again. This makes it a bit harder to distinguish the PuTTY windows on my desktop to find the correct host to work with.

### What do you expect to happen?

Either don't change that title at all, or at least change it back to what it was before when leaving the program.

### Please tell us about your environment:

- Node-RED version: v1.3.5
- Node.js version: v14.17.2
- npm version: 6.14.13
- Platform/OS: Raspberry Pi OS, Linux 5.10.17-v7+ arm LE
- Browser: n/a (PuTTY SSH terminal)

dceejay, 2021-07-10:
What does putty set it to ?

ygoe, 2021-07-11:
The host name that the session is connected to.

dceejay, 2021-07-11:
hmm - I don't have a windows box to test with putty at present- If I try a simple ssh session on Mac or Linux I see the title as
[image]
and when I then Ctrl-c to stop it I see
[image]
So it seems to revert fine in this instance.

ygoe, 2021-07-11:
Maybe your shell prompt does that? It has the current path in it after all. Mine doesn't touch the terminal title.

dceejay, 2021-07-11:
maybe it's a putty thing - and you can turn it off - see https://serverfault.com/questions/178469/how-can-i-set-puttys-window-title-to-the-name-of-the-loaded-session
[image]

ygoe, 2021-07-11:
I don't want to reconfigure my terminal just because a single program doesn't clean up after itself. Are you throwing away your stuff anywhere you go and then say: you could go another way so you don't see my garbage? I guess not. That's why I opened this issue to improve the situation. I'm not going to install Node-RED every week and it's really nothing critical, so this isn't an urgent issue.

dceejay, 2021-07-12:
Hi,
interesting point of view. I have shown you that a simple command line ssh session does restore the name ok, and that a 3rd party application over which I have no control decides not to, but can be set to ignore renaming - and somehow I'm the bad guy.

I have now found a windows 10 box with putty on and this is the result. (with default install of putty 1.7.5 - no special session config.)
And as you can see that also reverts the title just fine.

[image]

ygoe, 2021-07-12:
Please show me the value of your shell prompt variable, I think it's called `$PS1`. If it sets the terminal title, then that's your personal fun but not the default environment of at least Raspberry Pi OS, maybe many more. And if I want to be able to set the title from other programs, then configuring my terminal to (non-default, too) ignore the titles, that'll break. So after considering all the facts, yes, node-red has the fault.

dceejay, 2021-07-12:
Hi - I haven't knowingly touched this since installing Raspbian. How is yours set ?
```
pi@pi3:~ $ echo $PS1
\[\e]0;\u@\h: \w\a\]${debian_chroot:+($debian_chroot)}\[\033[01;32m\]\u@\h\[\033[00m\]:\[\033[01;34m\]\w \$\[\033[00m\]
```

dceejay, 2021-07-12:
just for grins I have also just tried a vanilla fedora system 
```
echo $PS1
[\u@\h \W]\$
```
and indeed putty reverts just fine for that also... so not sure where else to go with this.

ygoe, 2021-07-12:
```
echo $PS1
\n\A \[\e[0;31m\]\u\[\e[0m\]@\[\e[0;33m\]\h\[\e[0m\] \[\e[1;34m\]\w \[\e[0m\]>
```

Sorry, might be `$PROMPT_COMMAND`, mine is empty. I don't know much about these features of bash.

Well, if you can't reproduce this anywhere and nobody else is here to care, it must be my computer. I'll just add a note in my installation instructions to relogin after this.

dceejay, 2021-07-13:
my `$PROMPT_COMMAND` is also empty, but using your PS1 does also cause mine to fail... finally I can see your issue.