node-red/linux-installers #17: Install fails on AlmaLinux 8.4

andrewm659, 2021-08-19:
Trying to install nodered on Almalinux and get the following error:

```
Doesn't seem to be running on RedHat, Centos, Fedora, or Oracle Linux, so quitting
```
I tried removing the symlink from /etc/centos-release to /etc/almalinux-release and recreating the files with the CentOS and or Red Hat release info.  Still didn't work.

Is this resolved in version 16?

dceejay, 2021-08-19:
never heard of almalinux so never tested. Happy to consider a pull  request if anyone willing to work it out.

andrewm659, 2021-08-19:
Alma Linux and Rocky Linux are the new stable clones of RedHat and or CentOS.

dceejay, 2021-08-19:
If you grab the actual install script and edit line 67 to include the relevant release word - then what error do you get ?

andrewm659, 2021-08-19:
dnf.

andrewm659, 2021-08-19:
I made it look like this and it worked
```
versions='fedora"centos"rhel"ol"almalinux"rockylinux'
```

dceejay, 2021-08-20:
did you also need to change line 103 so it picks the dnf path ?

commit 6de41c04 (2021-08-23): update to include Rocky and Alma linux

to close #17
closed 2021-08-23 by dceejay