node-red/linux-installers #23: Update via parallel-ssh?

n3odon, 2022-01-11:
I am trying to update several units in parallel via ssh. I can run the script (here for debian), but the root passord is needed and it can't be passed via a pipe (I tryied echo 'password' | bash ... , and echo 'password' | sudo -S bash ...). 

Is there a way to pass the password or run the script without the need for a password? Or another way around this?

dceejay, 2022-01-11:
There is a new option --nodered-user that allows you to run the script as root and specify the node-red user that will be set as the service user. In theory you could then allow root to have ssh access  and only do the login once and then it shouldn't prompt  - but you have to weigh that against the risk of allowing root access over ssh.

n3odon, 2022-01-11:
I ended up using this option, yes. Here is the single line to update node-red for alle the hosts listed in the file hostnames:

parallel-ssh -i -h hostnames -o logfolder -l pi -A -t 0 'echo <sudopassword> | sudo -S bash node-red-update.sh --confirm-root --confirm-install --confirm-pi --update-nodes --nodered-user=pi'

I first downloaded the script via
parallel-ssh -i -h hostnames -o logfolder -l pi -A -t 0 'curl -sL https://raw.githubusercontent.com/node-red/linux-installers/master/deb/update-nodejs-and-nodered > node-red-update.sh'

Thank you for the confirmation, @dceejay !

closed 2022-01-11 by n3odon