node-red/linux-installers #25: Install script for Raspberry Pi overwrites custom nodered.service

ralphwetzel, 2022-01-15:
Hi there!
I used to have node-red 2.1.4 running on a RPi. Installation was done according to the [given instruction](https://nodered.org/docs/getting-started/raspberrypi). I've customized `/lib/systemd/system/nodered.service` to define some special parameters - e.g. `--userDir`.

Today I just upgraded this installation - again following the [instruction given](https://nodered.org/docs/getting-started/raspberrypi) - to node-red 2.1.5.

After successfull completion of the process I discovered, that my customized `/lib/systemd/system/nodered.service` was overwritten.

Proposal: Only create `nodered.service`, if it's missing - but do not overwrite it.
Edge case: If there's a breaking change in `nodered.service`, that's mandatory to be installed - backup the current file & raise a supportive comment during installation.

dceejay, 2022-01-15:
sensible suggestion.

ralphwetzel, 2022-01-15:
:+1: I'll propose a PR.