node-red/linux-installers #27: RPI install script wrong flag management

FStefanni, 2022-02-01:
Hi,

I have successfully set up Node-RED as a systemd service on rpi 4 with NVM.
To do so, I used the Node-RED rpi [official install script](https://raw.githubusercontent.com/node-red/linux-installers/master/deb/update-nodejs-and-nodered) .

After performing a first run of the script, and having noticed that it was not working as expected (IMHO), I checked such a script and found that its `$GLOBAL` flag seems not set correctly.
In fact, it is initialized to `true`, and then it should be set to `false`, since I am using NVM, but this never happens.

So to make the correct installation, I just changed its initial value (I now this is not the correct fix, but the script does a lot of things, and I have no time to check them all).

This leaves the system with a local installation of Node-RED,  and a not working systemd init script (`nodered.service`). But then it is quite easy to make it work (if you are interested to what I have done to make it work, I'll post here the other performed stuff -- basically few simple changes to the `nodered.service` file).

Nevertheless, the value of the `$GLOBAL` flag seems incorrect to me, in case of NVM.

Regards

dceejay, 2022-02-02:
Hi @FStefanni - not quite sure what you are saying... the GLOBAL variable should get set false if you are using nvm on line 415 (after the if on line 405)  

So does it detect nvm (line 405 and take that branch - and eg echo line 410 - or does the test fail ? 

So yes please - I'm interested to try to fix this/ see what yo have done. Thanks

FStefanni, 2022-02-03:
Hi,

sorry I was not clear enough: I'll try to better clarify.
Basically, the `GLOBAL` variable is initialized to `true` in the script, and, as you say, at line 415 *should* become `false` .
But I put a `echo` just after the assignment (to check if was performed), and I saw no print at all, so I suppose the script takes another execution path... (probably is this the bug?).
So in my case, I did not performed any actual fix, and just initialized the `GLOBAL` to `false`, and the script worked.

Then, I have fixed the running with systemd and NVM.

Basically, the idea is simple: move the Node-RED execution call into a script, so we can load NVM before starting Node-RED.

The detailed steps are the following:
1. Create a small bash script `nodered-service.sh`:
```bash
#!/usr/bin/env bash
. /home/pi/.nvm/nvm.sh
/usr/bin/env node-red-pi $@
# EOF
```
2. Place the script anywhere you like, and make it executable: `chmod a+x nodered-service.sh`. In my case I placed it in `/home/pi`, but I believe that if automatically generated by the Node-RED install script, it should probably go into `/home/pi/.node-red`. Also the case of different user names should be probably handled (I have not checked if the install script supports other users than "pi").
3. Change the Node-RED systemd file `/lib/systemd/system/nodered.service`:
```bash
# Comment the following line:
# ExecStart=/usr/bin/env node-red-pi $NODE_OPTIONS $NODE_RED_OPTIONS
# Insert the following line. Please note to use the correct path to the script, and to pass the same options
# of the commented line
ExecStart=/home/pi/nodered-service.sh $NODE_OPTIONS $NODE_RED_OPTIONS
# Insert also this line. According with some quick search, some people had problems in killing 
# all processes without this.
# I have not checked if it still happens: I have just inserted this line for safety sake.
# Maybe a better than me systemd expert could clarify if it is really needed...
KillMode=control-group
```

That's all :)

Regards.

dceejay, 2022-02-03:
which user  are you running the script as ?  what does `echo $HOME` report ? does the  /home/{youruser}/.nvm  exist ?
(as in yes that test on line 405 should not fail).

FStefanni, 2022-02-04:
Hi,

> which user are you running the script as ? 

I suppose we are talking about the Node-RED install script.
I run it as user `pi`

> what does `echo $HOME` report ? 
```bash
> echo $HOME
/home/pi
```
> does the /home/{youruser}/.nvm exist ? (as in yes that test on line 405 should not fail).

Yes, it exists, and in fact the script prints "Using nvm is NOT RECOMMENDED. Node-RED will not run as a service under nvm".

Regards

closed 2022-11-09 by dceejay