node-red/linux-installers #31: Installer bash script fails to install node-red V3.0.0

juggledad, 2022-07-14:
when running the bash script on a Pi, the install fails (https://discourse.nodered.org/t/node-red-v3-0-0-pi-installer-script-node-red-does-not-start-after-installer-is-finished-setting-js-no-such-file-or-directory/65085/4)

However if you use `sudo npm install -g --unsafe-perm node-red` it does install

This is the nstall log using the bash script:
pi@testpi:~ $ cat /var/log/nodered-install.log
Already have nodejs v14.20.0
OLD nodejs 14 :
NEW nodejs  :

***************************************

Started : Thu 14 Jul 2022 06:59:41 AM EDT
Running for user pi at /home/pi
Found global nodes:   :

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Reading package lists...
Building dependency tree...
Reading state information...
Package 'nodered' is not installed, so not removed
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

Versions: node:v14.20.0 npm:6.14.17
Now install Node-RED 
npm WARN saveError ENOENT: no such file or directory, open '/home/pi/package.json'
npm WARN enoent ENOENT: no such file or directory, open '/home/pi/package.json'
npm WARN node-red-contrib-mp4frag@0.1.0-beta.4 requires a peer of socket.io@2 || 3 || 4 but none is installed. You must install peer dependencies yourself.
npm WARN node-red-contrib-ui-mp4frag@0.1.0-beta.3 requires a peer of node-red-dashboard@^2.26.1 but none is installed. You must install peer dependencies yourself.
npm WARN pi No description
npm WARN pi No repository field.
npm WARN pi No README data
npm WARN pi No license field.

+ node-red@3.0.0
updated 1 package in 15.789s
3.0.0 
Now create basic package.json for the user and move any global nodes
Installing extra nodes: node-red-node-pi-gpio@latest node-red-node-random@latest node-red-node-ping@latest node-red-contrib-play-audio@latest node-red-node-smooth@latest node-red-node-serialport@latest :
npm WARN expo-random@12.1.2 requires a peer of expo@* but none is installed. You must install peer dependencies yourself.
npm WARN node-red-contrib-ui-mp4frag@0.1.0-beta.3 requires a peer of node-red-dashboard@^2.26.1 but none is installed. You must install peer dependencies yourself.
npm WARN react-native-securerandom@0.1.1 requires a peer of react-native@* but none is installed. You must install peer dependencies yourself.

+ node-red-node-random@0.4.0
+ node-red-node-ping@0.3.1
+ node-red-contrib-play-audio@2.5.0
+ node-red-node-smooth@0.1.2
+ node-red-node-serialport@1.0.1
+ node-red-node-pi-gpio@2.0.4
updated 6 packages in 28.993s
Now add the shortcut and start/stop/log scripts to the menu
Now add systemd script and configure it for pi:pi @ /home/pi
ln: failed to create symbolic link '/usr/bin/python': File exists
Now add launcher to top bar, add cpu temp example, make sure ping works
The user `pi' is already a member of `gpio'.

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Reading package lists...
Building dependency tree...
Reading state information...
python3-rpi.gpio is already the newest version (0.7.0-0.1~bpo10+4).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Memory  : 971M
Started :  Thu 14 Jul 2022 06:59:41 AM EDT 
Finished:  Thu 14 Jul 2022 07:01:00 AM EDT
pi@testpi:~ $

Steve-Mcl, 2022-07-14:
OK, I think i know what the issue is (but not certain on the "right" solution)

In simple terms, i beleieve this will work if you first update NPM

That is due to usage of `--location=global` which will not work on NPM v7 or older.

I know `--global` was depreciated in a single version of NPM v8.x? but was swiftly reverted in an immediate follow up version.

---

Potential solutions
1. stop install if NPM < 8
2. revert to using `-g` 
3. add both flags? `-g --location=global` ? (never tried)

dceejay, 2022-07-14:
happy to go back to -g... one sec
pushed

dceejay, 2022-07-14:
yes exactly - messes up the "ui" - but I guess if they have reverted it - it should clean up after a while when newer npm gets baked in.

closed 2022-07-14 by dceejay