node-red/linux-installers #35: script with no sudo

syswombat, 2022-11-02:
hello
would be nice to have the script running without sudo 

/dev/fd/63: Zeile 278: sudo: Kommando nicht gefunden.
/dev/fd/63: Zeile 281: sudo: Kommando nicht gefunden.
/dev/fd/63: Zeile 283: sudo: Kommando nicht gefunden.
/dev/fd/63: Zeile 284: sudo: Kommando nicht gefunden.
/dev/fd/63: Zeile 293: sudo: Kommando nicht gefunden.

dceejay, 2022-11-02:
What operating system and version are you using ? How do you install applications as a normal user without using Suso to give you admin privileges ?

dceejay, 2022-11-03:
We really don't recommend installing as root... but I have just uploaded a version of the script that hopefully may detect root and not call sudo...

commit 52250ca7 (2022-11-03): bump non-supported version of node, detect running as root

and if so don't use sudo
to close #35
closed 2022-11-03 by dceejay