node-red/linux-installers #37: update-nodejs-and-nodered messes up filesystem rights when --node16 or --node18 provided on armv6l

friebi, 2022-11-08:
## Scenario
Trying to update NodeJS to v16 or v18 and Node-Red on a Raspberry Pi Zero W.
Command line given:
```
update-nodejs-and-nodered --confirm-install --confirm-pi --node18
```

## Expectation
NodeJS and Node-Red is updated without issue. Especially the script should not mess around using root rights.

## Observation
During installation process in the NodeJs state, the filesystem rights in `/usr/bin/*` are changed from `root:root` to `pi:pi`.

The script works fine, when `--node14` is provided, but does not work with `--node16` or `--node18` is provided.

That should not happen and in particular leads to a half broken system, becauso also the `sudo` executable is not owed by root anymore and any call with sudo will lead to the following error message:
```
sudo: /usr/bin/sudo must be owned by uid 0 and have the setuid bit set
```
I could only recover the functionality by mounting the SD card into another linux system and chown all files to `root` again.

I tried to debug and understand the script line by line. I could not identify the culprit line. I tried to execute the particular steps for installing nodeJS with the commands from the scripts manually.
```
curl -sSL -o /tmp/node.tgz https://unofficial-builds.nodejs.org/download/release/v18.12.0/node-v18.12.0-linux-armv6l.tar.gz
mkdir -p /tmp/nodejs
tar -zxof /tmp/node.tgz --strip-components=1 -C /tmp/nodejs
sudo chown -R 0:0 /tmp/nodejs
sudo cp -PR /tmp/nodejs/* /usr/
sudo rm -rf /tmp/nodejs
rm /tmp/node.tgz

# finally
update-nodejs-and-nodered --confirm-install --confirm-pi
```
Surprisingly this does work as expected and does not mess up the rights.

So unfortunately I could not nail it down to a particular line within the script, otherwise I had created a PR with a potential fix.

dceejay, 2022-11-08:
Was this using the update script from the web ie by running

```
bash <(curl -sL https://raw.githubusercontent.com/node-red/linux-installers/master/deb/update-nodejs-and-nodered)
```
or by running a local copy of the script ?  IE just `update-nodejs-and-nodered`  ?

If the latter then yes there was a bug in an older version of the script that has been fixed already in the online version.

friebi, 2022-11-08:
Thanks for the quick responce. I used the online version as of 06.11.2022. I always make sure to have the latest before executing.

dceejay, 2022-11-08:
First fixed back end of last year - https://github.com/node-red/linux-installers/commit/29d38f7eedae80b2b27254977a3383b66767181d

latest 4 days ago as part of the forum discussion - https://github.com/node-red/linux-installers/commit/f4017226e75c990a07bf4b37b24ad9c33efabff4

dceejay, 2022-11-09:
I have tried to re-create this using a Pi model 1 (Armv6) - installed old node12 and Node-RED v1 - then tried the upgrade script as-is and it seemed to work ok... no changing for permissions... then repeated with `--node18` added... again so far so good - all files in /usr/bin still owned by root. 

I don't doubt there _was_ a failure mode... so there could still be.. but I'm not sure how to provoke it.

friebi, 2022-11-11:
Sorry for late reply. I will test once more to be sure that I really used the latest version of the script. I will report back and close the issue if I cannot reproduce it anymore.

friebi, 2022-12-03:
Sorry, I was busy with so many other things until now.

I now tried
```
curl -sL https://raw.githubusercontent.com/node-red/linux-installers/master/deb/update-nodejs-and-nodered | bash -s -- --confirm-install --confirm-pi --node18 --restart
```
And this worked without any issue. Not sure what I have done wrong previously.
[image]

Thanks a lot for the supprt and I'm closing the issue 👍

closed 2022-12-03 by friebi