node-red/linux-installers #39: Install fails, finds no npm

jpelletier, 2023-05-27:
Tried to install node-red on a Banana pi M2 Zero with these instructions:
https://nodered.org/docs/getting-started/raspberrypi
The script fails, can't find npm

jpelletier, 2023-05-27:
Here is the log file:
[nodered-install.log](https://github.com/node-red/linux-installers/files/11581799/nodered-install.log)

Steve-Mcl, 2023-05-27:
Did you add `--node16` or `--node18` to the end of the command?

e.g.

```bash
bash <(curl -sL https://raw.githubusercontent.com/node-red/linux-installers/master/deb/update-nodejs-and-nodered) --node16
```

jpelletier, 2023-05-27:
I tried with --node16 at the end, as you mentioned. It worked with the Raspberry Pi but not the Banana Pi.

dceejay, 2023-05-28:
from your log - 
```
Err:3 http://mirror.albony.xyz/armbian jammy InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 93D6889F9F0E78D5
Reading package lists...
E: Repository 'https://itai-nelken.github.io/weekly-box86-debs/debian  InRelease' changed its 'Origin' value from 'weekly_box86_debs' to ''
E: Repository 'https://itai-nelken.github.io/weekly-box86-debs/debian  InRelease' changed its 'Label' value from 'weekly_box86_debs' to ''
E: Repository 'https://itai-nelken.github.io/weekly-box86-debs/debian  InRelease' changed its 'Codename' value from 'buster' to ''
W: GPG error: https://cli.github.com/packages stable InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 23F3D4EA75716059
E: The repository 'http://cli.github.com/packages stable InRelease' is not signed.
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: http://mirror.albony.xyz/armbian jammy InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 93D6889F9F0E78D5
Error executing command, exiting

```
So it seems that the Arabian repository it is trying to use for nodes isn't fully "signed" so nodejs is failing to install correctly (so then the script fails and then install node 12 from the default repo which doesn't contain npm and etc etc... not good.

I think you are going to have to install nodejs 16 manually first  - or get the correct signed key for that repo - and then you can run the script to install Node-RED on top.

jpelletier, 2023-07-17:
Update: I switched to the Raspberry Pi Zero 2 W because of the camera port, and I don't need node-red finally.