node-red/linux-installers #40: Debian installer not working correctly with bookworm

klein0r, 2023-06-27:
- Upgraded Raspberry Pi OS to **Debian 12 (bookworm)**
- Used the installer to install NodeRED and nodejs without any parameters
- Installer plans to install nodejs 16
- nodejs 18 is installed instead
- Install fails

[image]

[image]

## System info

```
pi@smarthomeraspi:~ $ lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 12 (bookworm)
Release:	12
Codename:	bookworm
```

## Log

```
***************************************

Started : Di 27. Jun 08:38:41 CEST 2023
Running for user pi at /home/pi
Failed to stop nodered.service: Unit nodered.service not loaded.
Found global nodes:   :

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Paketlisten werden gelesen…
Abhängigkeitsbaum wird aufgebaut…
Statusinformationen werden eingelesen…
E: Paket nodered kann nicht gefunden werden.
Installing nodejs 16

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Paketlisten werden gelesen…
Abhängigkeitsbaum wird aufgebaut…
Statusinformationen werden eingelesen…
Paket »nodejs-legacy« ist nicht installiert, wird also auch nicht entfernt.
Paket »nodejs« ist nicht installiert, wird also auch nicht entfernt.
Paket »npm« ist nicht installiert, wird also auch nicht entfernt.
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.
dpkg: Warnung: Die Anforderung, nodejs zu entfernen, wird ignoriert; es ist nicht installiert
dpkg: Warnung: Die Anforderung, node zu entfernen, wird ignoriert; es ist nicht installiert

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Paketlisten werden gelesen…
Abhängigkeitsbaum wird aufgebaut…
Statusinformationen werden eingelesen…
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.
Grab the LTS bundle

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Paketlisten werden gelesen…
Abhängigkeitsbaum wird aufgebaut…
Statusinformationen werden eingelesen…
curl ist schon die neueste Version (7.88.1-10).
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.

## Installing the NodeSource Node.js 16.x repo...

## Populating apt-get cache...

+ apt-get update
OK:1 http://security.debian.org/debian-security bookworm-security InRelease
OK:2 http://deb.debian.org/debian bookworm InRelease
OK:3 http://deb.debian.org/debian bookworm-updates InRelease
OK:4 http://archive.raspberrypi.org/debian bookworm InRelease
Paketlisten werden gelesen…

## Confirming "bookworm" is supported...

+ curl -sLf -o /dev/null 'https://deb.nodesource.com/node_16.x/dists/bookworm/Release'

## Adding the NodeSource signing key to your keyring...

+ curl -s https://deb.nodesource.com/gpgkey/nodesource.gpg.key | gpg --dearmor | tee /usr/share/keyrings/nodesource.gpg >/dev/null

## Creating apt sources list file for the NodeSource Node.js 16.x repo...

+ echo 'deb [signed-by=/usr/share/keyrings/nodesource.gpg] https://deb.nodesource.com/node_16.x bookworm main' > /etc/apt/sources.list.d/nodesource.list
+ echo 'deb-src [signed-by=/usr/share/keyrings/nodesource.gpg] https://deb.nodesource.com/node_16.x bookworm main' >> /etc/apt/sources.list.d/nodesource.list

## Running `apt-get update` for you...

+ apt-get update
OK:1 http://security.debian.org/debian-security bookworm-security InRelease
OK:2 http://deb.debian.org/debian bookworm InRelease
OK:3 http://deb.debian.org/debian bookworm-updates InRelease
OK:4 http://archive.raspberrypi.org/debian bookworm InRelease
Holen:5 https://deb.nodesource.com/node_16.x bookworm InRelease [4.586 B]
Holen:6 https://deb.nodesource.com/node_16.x bookworm/main armhf Packages [785 B]
Holen:7 https://deb.nodesource.com/node_16.x bookworm/main arm64 Packages [778 B]
Es wurden 6.149 B in 2 s geholt (3.869 B/s).
Paketlisten werden gelesen…

## Run `sudo apt-get install -y nodejs` to install Node.js 16.x and npm
## You may also need development tools to build native addons:
     sudo apt-get install gcc g++ make
## To install the Yarn package manager, run:
     curl -sL https://dl.yarnpkg.com/debian/pubkey.gpg | gpg --dearmor | sudo tee /usr/share/keyrings/yarnkey.gpg >/dev/null
     echo "deb [signed-by=/usr/share/keyrings/yarnkey.gpg] https://dl.yarnpkg.com/debian stable main" | sudo tee /etc/apt/sources.list.d/yarn.list
     sudo apt-get update && sudo apt-get install yarn

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Paketlisten werden gelesen…
Abhängigkeitsbaum wird aufgebaut…
Statusinformationen werden eingelesen…
Die folgenden zusätzlichen Pakete werden installiert:
  libc-ares2 libnode108 libuv1 node-acorn node-busboy node-cjs-module-lexer
  node-undici node-xtend nodejs-doc
Vorgeschlagene Pakete:
  npm
Die folgenden NEUEN Pakete werden installiert:
  libc-ares2 libnode108 libuv1 node-acorn node-busboy node-cjs-module-lexer
  node-undici node-xtend nodejs nodejs-doc
0 aktualisiert, 10 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.
Es müssen 13,7 MB an Archiven heruntergeladen werden.
Nach dieser Operation werden 64,7 MB Plattenplatz zusätzlich benutzt.
Holen:1 http://deb.debian.org/debian bookworm/main arm64 libc-ares2 arm64 1.18.1-3 [99,2 kB]
Holen:2 http://deb.debian.org/debian bookworm/main arm64 libuv1 arm64 1.44.2-1 [133 kB]
Holen:3 http://deb.debian.org/debian bookworm/main arm64 node-xtend all 4.0.2-3 [3.932 B]
Holen:4 http://deb.debian.org/debian bookworm/main arm64 nodejs arm64 18.13.0+dfsg1-1 [304 kB]
Holen:5 http://deb.debian.org/debian bookworm/main arm64 node-acorn all 8.8.1+ds+~cs25.17.7-2 [128 kB]
Holen:6 http://deb.debian.org/debian bookworm/main arm64 node-cjs-module-lexer all 1.2.2+dfsg-5 [30,1 kB]
Holen:7 http://deb.debian.org/debian bookworm/main arm64 node-busboy all 1.6.0+~cs2.6.0-2 [16,9 kB]
Holen:8 http://deb.debian.org/debian bookworm/main arm64 node-undici all 5.15.0+dfsg1+~cs20.10.9.3-1 [283 kB]
Holen:9 http://deb.debian.org/debian bookworm/main arm64 libnode108 arm64 18.13.0+dfsg1-1 [9.366 kB]
Holen:10 http://deb.debian.org/debian bookworm/main arm64 nodejs-doc all 18.13.0+dfsg1-1 [3.358 kB]
Es wurden 13,7 MB in 1 s geholt (11,0 MB/s).
Vormals nicht ausgewähltes Paket libc-ares2:arm64 wird gewählt.
(Lese Datenbank ... 41930 Dateien und Verzeichnisse sind derzeit installiert.)
Vorbereitung zum Entpacken von .../0-libc-ares2_1.18.1-3_arm64.deb ...
Entpacken von libc-ares2:arm64 (1.18.1-3) ...
Vormals nicht ausgewähltes Paket libuv1:arm64 wird gewählt.
Vorbereitung zum Entpacken von .../1-libuv1_1.44.2-1_arm64.deb ...
Entpacken von libuv1:arm64 (1.44.2-1) ...
Vormals nicht ausgewähltes Paket node-xtend wird gewählt.
Vorbereitung zum Entpacken von .../2-node-xtend_4.0.2-3_all.deb ...
Entpacken von node-xtend (4.0.2-3) ...
Vormals nicht ausgewähltes Paket nodejs wird gewählt.
Vorbereitung zum Entpacken von .../3-nodejs_18.13.0+dfsg1-1_arm64.deb ...
Entpacken von nodejs (18.13.0+dfsg1-1) ...
Vormals nicht ausgewähltes Paket node-acorn wird gewählt.
Vorbereitung zum Entpacken von .../4-node-acorn_8.8.1+ds+~cs25.17.7-2_all.deb ...
Entpacken von node-acorn (8.8.1+ds+~cs25.17.7-2) ...
Vormals nicht ausgewähltes Paket node-cjs-module-lexer wird gewählt.
Vorbereitung zum Entpacken von .../5-node-cjs-module-lexer_1.2.2+dfsg-5_all.deb ...
Entpacken von node-cjs-module-lexer (1.2.2+dfsg-5) ...
Vormals nicht ausgewähltes Paket node-busboy wird gewählt.
Vorbereitung zum Entpacken von .../6-node-busboy_1.6.0+~cs2.6.0-2_all.deb ...
Entpacken von node-busboy (1.6.0+~cs2.6.0-2) ...
Vormals nicht ausgewähltes Paket node-undici wird gewählt.
Vorbereitung zum Entpacken von .../7-node-undici_5.15.0+dfsg1+~cs20.10.9.3-1_all.deb ...
Entpacken von node-undici (5.15.0+dfsg1+~cs20.10.9.3-1) ...
Vormals nicht ausgewähltes Paket libnode108:arm64 wird gewählt.
Vorbereitung zum Entpacken von .../8-libnode108_18.13.0+dfsg1-1_arm64.deb ...
Entpacken von libnode108:arm64 (18.13.0+dfsg1-1) ...
Vormals nicht ausgewähltes Paket nodejs-doc wird gewählt.
Vorbereitung zum Entpacken von .../9-nodejs-doc_18.13.0+dfsg1-1_all.deb ...
Entpacken von nodejs-doc (18.13.0+dfsg1-1) ...
node-cjs-module-lexer (1.2.2+dfsg-5) wird eingerichtet ...
libc-ares2:arm64 (1.18.1-3) wird eingerichtet ...
libuv1:arm64 (1.44.2-1) wird eingerichtet ...
nodejs-doc (18.13.0+dfsg1-1) wird eingerichtet ...
node-xtend (4.0.2-3) wird eingerichtet ...
node-busboy (1.6.0+~cs2.6.0-2) wird eingerichtet ...
node-undici (5.15.0+dfsg1+~cs20.10.9.3-1) wird eingerichtet ...
node-acorn (8.8.1+ds+~cs25.17.7-2) wird eingerichtet ...
libnode108:arm64 (18.13.0+dfsg1-1) wird eingerichtet ...
nodejs (18.13.0+dfsg1-1) wird eingerichtet ...
update-alternatives: /usr/bin/nodejs wird verwendet, um /usr/bin/js (js) im automatischen Modus bereitzustellen
Trigger für man-db (2.11.2-2) werden verarbeitet ...
Trigger für libc-bin (2.36-9) werden verarbeitet ...

Versions: node:v18.13.0 npm:missing
Already have nodejs v18.13.0
OLD nodejs 18 :
NEW nodejs 18 :
```

klein0r, 2023-06-27:
Maybe an issue of the nodejs installer: https://github.com/nodesource/distributions/issues/1583

klein0r, 2023-06-27:
> Maybe an issue of the nodejs installer: [nodesource/distributions#1583](https://github.com/nodesource/distributions/issues/1583) or https://github.com/nodesource/distributions/issues/1579

> The challenge is that Debian Bookworm provides a newer Node.js package 18.13.0 from the ﻿deb.debian.org repository

Sounds like a great moment to set nodejs 18 as default in the node red installer?

dceejay, 2023-07-15:
Has RaspiOS released a version on bookworm yet ? (or did you do a manual upgrade ?)

klein0r, 2023-07-17:
@dceejay I've upgraded my Raspberry Pi 4 manually to Bookworm. But this is an issue on every debian installation

dceejay, 2023-07-17:
bookworm has it's own version of node18 in it's default repo - so although we ass the nodesource repo to the sources it then default to "latest" so picks the node18 version... and the debian one does not include npm and so it all goes bad.  

Also as noted RaspiOS has not yet moved to bookworm so yes while it's an issue for those that manually upgrade or install from debian direct it shouldn't bite everyone right now.

The proper fix as you suggest is to move our default to node18 and indeed I am testing that as we speak - but was hoping to release that along with the next version of core Node-RED (3.1)...

ampledata, 2023-07-21:
Just to note the work-around for Debian bookworm users, if you're installing using the bash+curl script (see: https://nodered.org/docs/getting-started/raspberrypi ) run the update-nodejs-and-nodered installer as follows:

```bash
bash <(curl -sL https://raw.githubusercontent.com/node-red/linux-installers/master/deb/update-nodejs-and-nodered) --node18
```

Tested on:
```
Distributor ID: Debian
Description:    Debian GNU/Linux 12 (bookworm)
Release:        12
Codename:       bookworm
```

dceejay, 2023-09-01:
latest version of script now defaults to 18 so should now work ok

closed 2023-09-01 by dceejay

mugginsjm, 2023-12-31:
5 attempts now... Clean install of "Bookworm" on RPI4 using Raspberry Pi Imager. Then tried running install using script with and without --node18 suffix.
Same error as klein0r in first post.
No idea what to do now.
Any help please

klein0r, 2024-01-01:
> No idea what to do now.

Just add the `--node18` suffix and specify a specific version.

mugginsjm, 2024-01-01:
Hi. content is:
deb [signed-by=/etc/apt/keyrings/nodesource.gpg] https://deb.nodesource.com/node_18.x nodistro main

I have tried with suffix --node18

thanks

dceejay, 2024-01-02:
script was updated yesterday (1/Jan/24) - should now be fixed.

mugginsjm, 2024-01-02:
That's great... working now
thanks