node-red/linux-installers #48: failing to install on GCP VM w/ Debian

nycalex, 2024-03-02:
hello

i am trying to install on debian 12 (bookwarm) in GCP cloud, using the script from https://nodered.org/docs/getting-started/raspberrypi

script fails to detect that i have sudo rights (I do).

For my VM instances I use  native google auth for  SSH, meaning i authenticate to VM SSH with my gmail credentials.

what i see in my /etc/sudoers:

# Allow members of group sudo to execute any command
%sudo   ALL=(ALL:ALL) ALL

when i run `sudo cat /etc/group | grep sudo`     i see:
sudo:x:27:
google-sudoers:x:1000:

so i don't clearly see how they integrate google auth into sudo architecture... but as a result your script fails

P.S. i have Node21 installed

THANKS for looking into it

dceejay, 2024-03-03:
What does
`groups "$USER"`
return ?

nycalex, 2024-03-03:
I created a workaround with a local non-google user, gave him sudoer rights and got through the installation. I think i'm ok for now. Thanks

dceejay, 2024-03-03:
ok - but would be good to know as the test I do is against the groups and not the sudoers files

nycalex, 2024-03-03:
Sorry, i deleted the environment yesterday after i finished playing with it.

closed 2024-03-04 by dceejay